kotelnik/plasma-applet-thermal-monitor #3: Empty Resources list

fatino, 2015-06-20:
Hi I'm on arch. Installed from aur. lm_sensors installed but resources list is empty.
This is my sensors output http://pastebin.com/ydKziawS

kotelnik, 2015-06-20:
Hi! That's weird. So when you open Thermal Monitor Settings... and in the first (Temperature) section under Resources table when you click on "+", the selectbox is entirely empty? If it is the case, please post somewhere log, which you get this way:
- open terminal
- killall plasmashell
- plasmashell &
Maybe there is a better way to aquire this log but currently I don't know about it...

fatino, 2015-06-20:
Ahhh that's how it works. I thought I was meant to select from resource list.
You should change the behaviour so when nothing is added there is big + sign or something. 
Kinda misleading this way. No way to know I should press + button.

kotelnik, 2015-06-20:
You're right, it isn't quite obvious when the list is empty... Thanks for bringing it up. Should be fixed now...

fatino, 2015-06-20:
Slowly becoming your UI guy xd JK

fatino, 2015-06-20:
00

closed 2015-06-20 by fatino

kotelnik, 2015-06-20:
Sure :-)