kotelnik/plasma-applet-thermal-monitor #4: lmsensors/asus-isa-0000/temp1 is not updated after resume from suspend

FadeMind, 2015-07-04:
After select watching lmsensors/asus-isa-0000/temp1 sensors applet working fine - updating temp. After suspend device and resume ISA sensor is not readable by applet and showing 0 (zero) value. 

Command sensors showing all sensors all time:

[tomasz@arch ~]$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +56.0°C  (high = +87.0°C, crit = +105.0°C)
Core 0:         +51.0°C  (high = +87.0°C, crit = +105.0°C)
Core 1:         +52.0°C  (high = +87.0°C, crit = +105.0°C)
Core 2:         +50.0°C  (high = +87.0°C, crit = +105.0°C)
Core 3:         +54.0°C  (high = +87.0°C, crit = +105.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:        +55.0°C  (crit = +108.0°C)

asus-isa-0000
Adapter: ISA adapter
temp1:        +55.0°C  

Workaround is logout and login again, after that ISA is updateable .

kotelnik, 2015-07-07:
That never happened to me but I have an idea how to fix it. My suspicion is that resources are not available immediately after resume so asking for them fails and it does not get fixed after plasma restart. I'll try to address this issue. Could you please try a little test? After the resume (and ISA is not updated anymore) go to widget settings and re-add the resource. Is the resource present? Does this workaround also repair the issue?

Thanks for reporting btw!

FadeMind, 2015-07-07:
Nope. This workaround not working at all. Only logout > login again fix this. 
[image]
[image]

FadeMind, 2015-07-07:
UPDATE:
ASUS devices have weird issue after resume. 

In default LED WIFI is blink if rfkill have no hard/soft block for WLAN and BT devices. And should working fine without any user action. 

    [tomasz@arch ~]$ modinfo asus_nb_wmi
    filename:       /lib/modules/4.1.1-1-mainline/kernel/drivers/platform/x86/asus-nb-wmi.ko.gz
    alias:          wmi:0B3CBB35-E3C2-45ED-91C2-4C5A6D195D1C
    license:        GPL
    description:    Asus Notebooks WMI Hotkey Driver
    author:         Corentin Chary <corentin.chary@gmail.com>
    depends:        asus-wmi
    intree:         Y
    vermagic:       4.1.1-1-mainline SMP preempt mod_unload modversions 
    parm:           wapf:WAPF value (uint)

dmesg (without useless [date] collumn:

    asus_wmi: Initialization: 0x1
    asus_wmi: BIOS WMI version: 7.9
    asus_wmi: SFUN value: 0x4a0877
    input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input41
    asus_wmi: Backlight controlled by ACPI video driver

SITUATION

After resume from suspend asus_nb_wmi module resume state of wlan and bt device based on rfkill and turn on Fn keys functionality but not turn ON led WLAN status. Only unload and load again module asus_nb_wmi FIX this issue. 

plasma-applet-thermal-monitor watching on active (loaded) modules and refresh status in interval. When some module is unload and applet refreshing status of temperature application freak out beacuse don't reload modules in meantime. 

After second unload and load asus_nb_wmi driver plasma-applet-thermal-monitor see again ASUS ISA sensor...

kotelnik, 2015-07-09:
Thanks for extended info. So if I understand correctly when you resume your PC you can see 0° on your ISA monitor. When you unload and load asus_nb_wmi module twice -> the temperature starts showing up again? (No other interaction with widget is needed?)

I'm also interested about the little test I requested from you and you did. Have you also tried to remove the ISA resource, then Apply changes and then add the resource again (and apply changes again) or did you only remove and immediately add the ISA resource? Please try the first mentioned if you haven't already done that. Thanks!

FadeMind, 2015-07-09:
Correct, but like I said: I made Your test with removing and adding new ISA in to panel. Not working at all.

kotelnik, 2015-07-10:
OK thanks. In that case it more likely is issue in plasma itself. My widget is using 'systemmonitor' data-engine of plasma which has named sources like the 'lmsensors/asus-isa-0000/temp1' of yours and many others. After reboot this source is still available (because you managed to re-add it) but it doesn't get updated anymore. I'm afraid I cannot do anything about it in my widget code...

closed 2016-04-05 by FadeMind