kotelnik/plasma-applet-thermal-monitor #6: Use UDisks2 instead of hddtemp?

Lekensteyn, 2015-08-23:
The netcat hack (not documented in README) is a bit ugly:
https://github.com/kotelnik/plasma-applet-thermal-monitor/blob/b188acddc7b6515a1ba4af3b1b3f56327ada51d3/package/contents/ui/config/ConfigTemperatures.qml#L246

For mobile devices it is also not nice to wake up every second. Consider using UDisks2 for reading the temperature. Its D-Bus interface is documented here: http://udisks.freedesktop.org/docs/latest/gdbus-org.freedesktop.UDisks2.Drive.Ata.html#gdbus-property-org-freedesktop-UDisks2-Drive-Ata.SmartTemperature. Advantage of using this interface is that it will not spin disks or wake up SSDs unless it is active.

kotelnik, 2015-08-24:
Hi! Thank you very much for your interest, suggestion and documentation. I didn't know about this interface and just ported the old widget. This is a great idea, I'll try to make it right.

kotelnik, 2015-08-30:
It was a bitch (and fun, too) but I finally learned a little about dbus interfaces and implemented the udisks2 one for querying the HDD temperature. Please let me know it it is working ok for you. And thanks again for the suggestion and documentation :)

closed 2015-09-10 by kotelnik