kotelnik/plasma-applet-thermal-monitor #7: HDD sources not properly initialized on plasma start

kotelnik, 2015-09-06:
Since Plasma upgraded to 5.4 HDD temp values are always at 0 when I reboot or logout/login or restart plasma shell. Then I have to remove HDD entries in Thermal Monitor settings and add them again, then it works fine. It's not something related to udisks2 change, it was like that with hdd-temp in previous version.

kotelnik, 2015-09-06:
I was unable to replicate this bug, yet. But it seams "executable" datasource is not ready on startup and therefore it is not properly initialized at the time connectedSources are filled. I'll do more testing and get back here with more info...

kotelnik, 2015-09-06:
Still unable to come up with something so I will need your (kajzer from aur:) help.

1)
While restarting plasma please see the plasma log - are there any messages containing "thermalMonitor"? Or you can just post me the whole log if you like.

2)
In the latest commit I've added "Reload Temperature Sources" action. This should skip the need for re-adding the HDD sources. When clicking this action I'd also like to see the plasma log.

Thank you!

kajzersoze, 2015-09-07:
"Reload Temperature Sources" fixed it, thanks.
I'll post log soon.

kajzersoze, 2015-09-08:
This is after login and HDD sources are at 0
http://pastebin.com/RfZwT6bh

and after "Reload Temperature Sources" :
http://pastebin.com/4tpVyrwX

kotelnik, 2015-09-10:
Thanks for the logs. I can see the same output I have. Though my HDD temperature gets always updated after a while... I'll add more debug logging and get back to you.

kotelnik, 2015-09-10:
I just did some minor adjustments and added more debug logging. Please give me the log after login again. And let the other temperatures update several times so I can see the updates in the log... Thanks!

kajzersoze, 2015-09-12:
Maybe it's just me, maybe something went wrong during plasma upgrade from 5.3 to 5.4, I'm not sure.
Anyway, if you keep "Reload Temperature Sources" option in future versions all is good ! :)

kotelnik, 2015-09-13:
Sure, that is the least I can do :)

kotelnik, 2015-10-18:
Hi! I finally encountered the issue you were having and possibly fixed it with the latest commit. Please try if it is fixed for you, too. Thanks!

kajzersoze, 2015-10-24:
Hi, sorry for late reply !
It's working fine now, cheers :)

kotelnik, 2015-10-24:
Glad to hear that :)

closed 2015-10-24 by kotelnik