kotelnik/plasma-applet-thermal-monitor #9: Freezed temp after switching off GPU with bbswith

BilyakA, 2015-11-10:
On Optimus graphics GPU temp sensor is active only when GPU is powered on.
On system bootup (when GPU is switched off) plasmoid showing 0. It's ok (though it can give something like "OFF" if can't read temp sensor). During using GPU it works perfectly. But after switching GPU off it freezes on last seen temp.

Way to reproduce:
1) take some notebook with Optimus graphics
2) install bumblebee and thermal monitor applet
3) switch on GPU
4) add GPU sensor to applet
5) turn off GPU

Result:
- applet shows last temp managed to read from sensor

Expected result:
- applet shows 0 or OFF

BilyakA, 2015-11-10:
OMG. While writing this issue plasmoid switched to 0 for GPU.
So maybe it is trying to read sensor and only after few attempts give up and show 0. It would be great to do this immediately (on next attempt to read data from sensor) right after switching off GPU so it could be controlled via applet.

kotelnik, 2015-11-12:
Hi! Thanks for reporting, Sorry for my late response. I'll try to address the issue. Unfortunately I cannot test it myself (I have no such PC with nvidia and intel) so I'll try to do a blind fix and add some logging and then I'll probably ask you for some logs.
I'll try to work on it next week.

BilyakA, 2015-11-12:
@kotelnik no problem

BilyakA, 2015-11-13:
@kotelnik This can be helpful:
eldorado@plumbum ~ $ nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

eldorado@plumbum ~ $ echo $?
9

I've tried to rewrite nvidiaDS in main.qml to handle 9 exit code and pass "OFF" or 0 but couldn't managed to. My QML skill is very close to 0, so I guess I can't help you with code.

kotelnik, 2015-11-13:
Nice! You've done more then enough for me to fix the issue. Can you please tell me if this command returned this error right away or after some time?

BilyakA, 2015-11-13:
@kotelnik it is returned immediately

kotelnik, 2015-11-13:
Sweet, I try to provide a fix tonight.

kotelnik, 2015-11-13:
Please try the latest git. If it starts to work as expected, I'll make a regular release...

BilyakA, 2015-11-13:
Fantastic! I've tested switching on and off GPU with and without overheat/meltdown indication. Everything works fine!

closed 2015-11-13 by BilyakA

kotelnik, 2015-11-13:
Thanks for testing :)