kotelnik/plasma-applet-thermal-monitor #10: Feature request: custom font settings

BilyakA, 2015-11-16:
Hi there!
Since github doesn't provide any kind of feature request page I'll post it here.
It would be great to have custom font settings for applet to use. I like a lot your warning and meltdown icons so I maximized their size in applet (to make them more visible in system tray). But as temp text is not bold I can hardly see it over colored icon. So it would be great to have font settings for label and temp text to make them bold/italic or to change font (not only text size).

Also a little issue easy to fix: warning and meltdown icons covers label text. I think it shouldn't.

kotelnik, 2015-11-17:
Yes, you're right, it doesn't look nice with the big icons under. So far I did the easy fix: label text is not covered by icon anymore. And I did another thing for readability - option for enabling drop shadows for label and temperature text. (latest git commit)

Of course I can still add the font options but I'll do that only if you still need it. So far it uses the default system-wide font...

RE: PS: I used to have gentoo, too :) So I remember the sleepless nights when compiling the whole KDE suite and other updates like firefox, chromium and stuff - it took ages... Then my fan started to be too lazy to manage so I switched to Arch :) But I still think it's one of the best distributions to have.

BilyakA, 2015-11-17:
I'll try new build only tomorrow (it is almost midnight). Will make you know what I think. Thanks for fast fixes and for great applet!

BilyakA, 2015-11-18:
Issue is closed. Thank you for fast fix!
Please remember to update kde-apps and kde-look applet to make it downloadable from plasma.

closed 2015-11-18 by BilyakA

kotelnik, 2015-11-18:
Sure, done :) np.