kotelnik/plasma-applet-thermal-monitor #11: "Reload Temperature Sources" button resets text and icon settings

BilyakA, 2015-11-19:
Hello again, my friend :)

I've noticed strange behavior of "Reload Temperature Sources". It resets layout and all text and icons settings. I'll investigate this and provide you normal bag report. Also I am using github version 89de935 and not that one from kde-apps. Try reproducing it yourself, if you won't be able to - i'll give a try again this afternoon.

kotelnik, 2015-11-19:
Hi! :) That's absolutely OK, it would be probably unrelated to the change anyway.

I cannot reproduce this behaviour. If you are using the git version, did you remove the kde-apps applet from ~/.local/share/plasma/plasmoids/ so there is only one applet source code in the system?

BilyakA, 2015-11-26:
@kotelnik Hi again!
I am very sorry for such a late response.
I can not reproduce this issue on latest build from kde-apps.
I guess I was my own installation issue or some problems related to previous installations.

closed 2015-11-26 by BilyakA

kotelnik, 2015-11-26:
I'm glad you resolved it :)