kotelnik/plasma-applet-thermal-monitor #12: Please support HiDPI

darkbasic, 2015-11-22:
[image]

kotelnik, 2015-11-22:
Hi! I can only fix the thermal monitor from the shown widgets. And I think hiDPI was already fixed in it. What needs fixing is desktop placement (so the text is scaling not only according the widget height). For now you can work around it by setting a proper widget hight and width with mouse.

darkbasic, 2015-11-22:
But if I do so it will look horrible when using the external standard DPI monitor :(

kotelnik, 2015-11-22:
I'm guessing you don't have the latest version (1.2.5 or git) because in the latest version the text in widget scales along with widget height no matter the font DPI. The mentioned fix for desktop placement (so the text is not overlapping the background) will be covered in next version.

darkbasic, 2015-11-22:
The screenshot is from git version: https://aur.archlinux.org/packages/plasma5-applets-thermal-monitor-git/

kotelnik, 2015-11-22:
Ok, I've just pushed the desktop placement fix. Now it looks like this on my desktop:
[image]
I probably don't know what you mean by the "horrible" look. What exactly should be different?

darkbasic, 2015-11-24:
Thanks, MUCH better but the applet size doesn't scale as expected.
[image]
[image]
[image]

kotelnik, 2015-11-24:
I still don't know what you mean or what exactly do you expect. You mean that the applet itself (the container) should automatically scale?

Or do you mean that the text should get bigger on hiDPI monitor and stay smaller on regular one? If this is the case it will not scale because the text size is only dependent on the applet size (and on configuration of the applet - Appearance section). And I implemented it this way so it looks still the same in panel with various height (or width in vertical one).

darkbasic, 2015-11-25:
With your latest version the text already gets bigger up to fit the container, the problem is the container itself which keeps a fixed pixel dimension, while it should double the pixel width and height when using an HiDPI monitor.

kotelnik, 2015-11-27:
OK I understand now, the container should get bigger, too. I'll see what I can do.

kotelnik, 2016-01-01:
Hi! I've finally made some time to look into this so let's make one more iteration :). Sorry for the delay.

Since I don't have a HiDPI monitor I've only tested with changing font dpi. I suspect it should mimic the HiDPI monitor but please correct me if I'm wrong.

I aligned the plasma built-in monitors (CPU, memory, network, system load) and mine (thermal monitor) the way you do. Then I changed dpi from 96 (my default) to e.g. 192. Some of the widgets got bigger because their minimal size changed according the dpi. They also get unaligned so I aligned them once again.

Then I changed back to 96 dpi. They stayed the same pixel size. Then I changed to 192 once again and they maintained the same pixel size.

So for me it looks like they behave the same as my widget. Only my widget has really small minimal size so it won't get bigger with bigger dpi.

But I didn't test with different screen resolution. So maybe combination of changing screen resolution and dpi can mimic attaching HiDPI monitor? Do you know how can I try and test it without this kind of monitor? I also cannot see what is wrong with the thermal widget from the last pictures you have attached here. On both monitors I can see the same thermal widget hight portion of system load viewer which I suppose is scaling fine.