kotelnik/plasma-applet-thermal-monitor #14: .xsession-errors gets spammed with qml print() output

Caltheron, 2015-12-21:
Qml seems to write every print() output into ~/.xsessio-errors which as a result gets spammed with the like of
<code>qml: New data incomming. Source: lmsensors/it8720-isa-0228/temp1, data: 32.000000</code>
<code>qml: updating model with temperature: 32, key: lmsensors/it8720-isa-0228/temp1, i: 0</code>

I'm not sure about how to fix it but would it help to use [QT's console API](https://doc.qt.io/qt-5/qtquick-debugging.html) instead?

kotelnik, 2015-12-21:
Interesting, I didn't know about qml flooding ~/.xsession-errors, thanks for the information! I'll investigate the Qt console API possibility. But in any case I should disable all logging when making a release commit. I'll do that soon.

kotelnik, 2016-01-01:
Hi! I've noticed now that my .xsession-errors log is no longer updated since late october (maybe because some KF 5.xx update?). So I cannot test if using "console.log" is avoiding the .xsession-errors or not. I can only tell that it is logging to the console as well as "print" function. That's what I've tested then.

Anyway I disabled logging in the latest release and I will only enable it locally in my repo for debugging purposes. Therefore there shouldn't be any more debug logging messages coming from this widget.