kotelnik/plasma-applet-thermal-monitor #16: Thermal monitor requires manual reload in every plasmashell restart

ongun-kanat, 2016-03-30:
When I start plasmashell thermal monitor does not automaticaly update itself. I have to right click and select reload every time I restarted plasmashell.

I'm using plasma 5.6 on Arch Linux

kotelnik, 2016-03-30:
Yes, I'm now also noticing this since I've updated my Arch... Thanks! Will try to fix that soon.

BilyakA, 2016-03-31:
4.1.12-gentoo with kde-plasma/plasma-meta-5.6.1 is here and no such problems.
Using last version of Thermal Monitor form Plasma Add-on installer

kotelnik, 2016-04-05:
Thanks for info. There really is a problem with property initialization - I believe I've repaired that in latest commits. Please re-test from git and let me know. If it is fine I'll create a new tag and post a .plasmoid on kde-apps. Thanks!

ongun-kanat, 2016-04-06:
Looks like no, it hasn't been resolved. I reinstalled from AUR git package.

kotelnik, 2016-04-10:
Thanks! I looked into the problem and it looks like it is a general problem of initialization of array and map properties in plasma widgets. They get re-initialized **several** times when plasma is starting but also e.g. when external monitor is connected / disconnected. It can (and will) be repaired on my (widget) side. But I consider it a bug in plasma 5.6 (or Qt 5.6, I'm not sure which one's fault is it) so I'll report it elsewhere.

Please wait for my second attempt to repair this issue, I'll do that soon and get back to You.

kotelnik, 2016-04-10:
Committed second attempt to fix this. Please try again :).

wolfyrion, 2016-04-11:
it seems that the one from AUR is working fine

kotelnik, 2016-04-12:
OK, thanks a lot, closing and making a tag.

mamrhein, 2016-07-17:
I'm running a laptop and an Intel NUC, both with KDE Frameworks 5.24 and KDE Plasma 5.7.1.
While on the laptop Thermal Monitor is displaying the temperature instantly after start, on the NUC I have to reload the temperature sources each time.

brigadier, 2016-08-03:
not resolved in KDE Plasma 5.7 from gentoo

kotelnik, 2016-08-07:
Alright, thanks for reports, I'll investigate further.

mamrhein, 2016-10-18:
With KDE Frameworks 5.27 and KDE Plasma 5.8.1 it does work on my Intel NUC too.
Thanks!

glaubersm, 2018-11-30:
Here cpu temperature is not shown immediately after login. I need to reload thermal monitor manually.

Operating System: Arch Linux 
KDE Plasma Version: 5.14.4
Qt Version: 5.12.0 RC2
KDE Frameworks Version: 5.52.0
Kernel Version: 4.19.4-arch1-1-ARCH

mamrhein, 2019-01-20:
After upgrading Qt and KDE, I have the same problem again. After start of Plasma CPU and MB temperature are not show, while disk temperature is.
Qt 5.12 / KDE Frameworks 5.54 / Plasma 5.14.4 / Thermal Monitor 1.2.8

roachsinai, 2019-01-21:
Who care about this problem maybe could goto #53 to check the reason(not sure).

artemanufrij, 2019-02-20:
Same here since a couple of weeks

andreimarinescu, 2019-02-21:
yeah, having the same issue. did anyone figure out a workaround yet?

roachsinai, 2019-02-21:
@andreimarinescu @artemanufrij #54 check it!

artemanufrij, 2019-02-21:
@roachsinai thank you for your feedback...

andreimarinescu, 2019-02-22:
@roachsinai I cloned your fork of the repo (the one used for the PR), followed the installation instructions and did a reboot. I still need to "Reload temperature sources" manually in order for the CPU temp to show up. Are there any other steps needed?

roachsinai, 2019-02-22:
Maybe you should read this issue #53 for some details.

beh666, 2019-12-29:
The problem still persists for me on two different computers. One running Arch and the other one Manjaro.

wolfyrion, 2019-12-29:
> The problem still persists for me on two different computers. One running Arch and the other one Manjaro.

for people who still have problems , simple solution

go -->

https://raw.githubusercontent.com/kotelnik/plasma-applet-thermal-monitor/b389b886a7ee759d06c9eb0e6a3b6abbea400035/package/contents/ui/main.qml 84

Copy Everything

edit

/home/your username/.local/share/plasma/plasmoids/org.kde.thermalMonitor/contents/ui/main.qml

Paste Everything

Save, Restart and you are done

beh666, 2020-01-04:
Thanks @wolfyrion! That worked perfectly!

Do you know the reason why this is not already implemented as default?

africsnail, 2020-02-29:
> Thanks @wolfyrion! That worked perfectly!
> 
> Do you know the reason why this is not already implemented as default?

Still isn't. At least on AUR

nagelp, 2020-06-07:
I also still have this problem on Ubuntu Linux, and the above-mentioned quick fix is working. Would I have gotten the fix if I had deleted the .local/share/plasma directory for my user? Don't want to try it now...

tomchiverton, 2020-07-31:
> https://raw.githubusercontent.com/kotelnik/plasma-applet-thermal-monitor/b389b886a7ee759d06c9eb0e6a3b6abbea400035/package/contents/ui/main.qml

Fixed here, Ubuntu 20.04. The diff for that is just :

```
--- /home/falken//.local/share/plasma/plasmoids/org.kde.thermalMonitor/contents/ui/main.qml     2020-07-31 17:28:25.498982256 +0100
+++ /home/falken//.local/share/plasma/plasmoids/org.kde.thermalMonitor/contents/ui/main.qml.dist        2020-07-31 17:27:56.842175969 +0100
@@ -190,10 +190,6 @@
 
         temperatureModel.clear()
 
-        if (!systemmonitorAvailableSources) {
-            systemmonitorAvailableSources = []
-        }
-
         if (!systemmonitorSourcesToAdd) {
             systemmonitorSourcesToAdd = []
         }
@@ -406,4 +402,3 @@
     }
 
 }
```