kotelnik/plasma-applet-thermal-monitor #20: Don't call nvidia-smi after resume from suspend on NVIDIA Optimus laptops

FadeMind, 2016-05-28:
@kotelnik 
plasma-applet-thermal-monitor can watch on NVIDIA GPU (via `nvidia-smi`).
Sadly after resume from suspend applet check status of `nvidia-smi` and turn on GPU and ignore bbswitch.
Reference: https://github.com/Bumblebee-Project/Bumblebee/issues/774

BilyakA, 2016-05-29:
So it is not bbswitch issue that after resuming from suspend GPU is turned on? 

I was trying to fix it in bbswitch settings with different versions of bbswitch and nvidia drivers for about a week before giving up -_-

Goddard, 2016-07-04:
I was trying to get nvidia gpu temp to show up.  I have a 970m.  Any ideas? Thanks

kotelnik, 2016-08-07:
Do you have "nvidia-smi" command available?

ArchangeGabriel, 2017-01-06:
It can also work with nouveau, but not on all cards I think. Not sure about 970m.