kotelnik/plasma-applet-thermal-monitor #22: Allow setting sensor as fan (=RPM)

z3ntu, 2016-09-01:
My laptop has the following sensor (gotten from `sensors`):
```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:     2800 RPM
temp1:        +70.0°C
```
but the plasmoid can only show temperature and cries because its 2800°C

kotelnik, 2017-08-20:
This widget was intended to show just temperatures. Working in RPM will need some work, maybe I do that some time :).

igorsantos07, 2017-09-08:
I'm on this one as well. I'll add here some suggested improvements to get it done:
- [ ] show custom unit, or hide unit for a given sensor
- [ ] allow for limit values with more than 2 chars (I can't set the warning to 3000 for instance)

acutbal, 2019-01-24:
I also think that would be very nice and useful if the widget could read the fan speed and display it as well. That could make this widget a universal measuring tool. 

Regards!! :smile:

dwardor, 2019-12-18:
Well I wanted it too so I did it... as a new applet heavily inspired/copied from thermal monitor
https://github.com/dwardor/plasma-applet-fanspeed-monitor