kotelnik/plasma-applet-thermal-monitor #30: Center temperature if alias is not set

andreyorst, 2017-07-30:
I just need to see my temperature and i do not care about what core it is, so i just dot set the `alias`. But this looks awful: 
[image]
It would be nucho more beautiful if temerature were centered vertically if alias don't set.
I can center it by increasing the font size, but this looks terrible too:
[image]
And Even if alias is set it is hard to read it in a nor,al size panel which make it even more useless

kotelnik, 2017-08-20:
Should now be in latest commits, thanks for suggestion :)! Please reopen if necessary.

closed 2017-08-20 by kotelnik

andreyorst, 2017-11-06:
Totally forgot to check. Well, maybe you misunderstood me, but I meant to center temperature should be centered vertically in the panel. No matter what font size do you use temperature still at the bottom.
I've installed it from git, to be sure.
[image]

and using slider to change the font size is kinda painful by the way
[image]