kotelnik/plasma-applet-thermal-monitor #41: System (journalctl) log is filled with repeated messages

eduardosm, 2018-03-27:
With this widget, system log flooded with the same message:
``plasmashell[962]: qml: temp unit: 0``

Probably it has something to do with this line: https://github.com/kotelnik/plasma-applet-thermal-monitor/blob/0002326c677391a93d30320bfda1de608aca1ee7/package/contents/code/temperature-utils.js#L16

eduardosm, 2018-04-09:
It seems that using dbgprint instead of print (last commit) fixes it, but has not been released into a version yet. Could you please release a version that includes that change?

keks24, 2018-10-30:
Even with this fix, it still spams `/home/ramon/.local/share/sddm/xorg-session.log`:
```
qml: temp unit: 0
```

```bash
ramon@lenovo ~ % plasmashell --version
plasmashell 5.13.5
ramon@lenovo ~ % plasmapkg2 --version
KCatalog being used without a Q*Application instance. Some translations won't work
kpackagetool5 2.0
ramon@lenovo ~ % uname -a
Linux lenovo 4.14.65-gentoo #1 SMP Mon Aug 20 08:12:06 CEST 2018 x86_64 Intel(R) Core(TM) i7-4700MQ CPU @ 2.40GHz GenuineIntel GNU/Linux
ramon@lenovo ~ %
```

Krovikan, 2019-03-04:
Any hope to fix this? Today, March 4th 2019 still have the journal flooded with:

plasmashell[XXX]: qml: temp unit: 0

XXX is a number that increase with the time

max0x7ba, 2020-03-27:
It still spams `~/.xsession-errors` with `qml: temp unit: 0`. Can it be fixed, please?

max0x7ba, 2020-03-27:
It still spams `.xsession-errors` with `qml: temp unit: 0` messages almost every second.

jolichter, 2020-04-05:
Hi, I've the same issue: journalctl log is filled with repeated messages: qml: temp unit: 0
As workaround I comment out this line in temperature-utils.js:
`    // print('temp unit: ' + temperatureUnit)`

my system Manjaro with KDE 5.18.3

nicolamori, 2020-04-06:
I have this issue too, on Archlinux with Plasma 5.18.4. I circumvented the problem by disabling the Qt debug logging by putting these:
```
QT_LOGGING_RULES='*.debug=false'
export QT_LOGGING_RULES
```
in `/etc/profile`, as suggested [here](https://forum.manjaro.org/t/kde-plasma-related-error-messages-in-journal/21790/4).

marjau, 2021-03-15:
I use thermalMonitorFix, i think @nicolamori workaround is less system intrusive

plasmashell 5.21.2

FILE: 
.local/share/plasma/plasmoids/org.kde.thermalMonitorFix/contents/code/temperature-utils.js

restart session, now spam go away, thks  @nicolamori