kotelnik/plasma-applet-thermal-monitor #47: Can not build Ubuntu 18

asmyshlyaev177, 2018-08-19:
When try to run `cmake .. -DCMAKE_INSTALL_PREFIX=/usr`

Got error

> CMake Error at CMakeLists.txt:6 (find_package):
>   Could not find a package configuration file provided by "ECM" (requested
>   version 0.0.11) with any of the following names:
> 
>     ECMConfig.cmake
>     ecm-config.cmake
> 
>   Add the installation prefix of "ECM" to CMAKE_PREFIX_PATH or set "ECM_DIR"
>   to a directory containing one of the above files.  If "ECM" provides a
>   separate development package or SDK, be sure it has been installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/alex/plasma-applet-thermal-monitor/build/CMakeFiles/CMakeOutput.log".

Log file https://ufile.io/6caj0
System - Ubuntu 18, 4.17.0-12.1-liquorix-amd64

Malekblubb, 2018-09-12:
You can fix this by simply installing "extra-cmake-modules":
arch: sudo pacman -S extra-cmake-modules
or
ubuntu: sudo apt-get install extra-cmake-modules

robertgroh, 2020-01-17:
I also needed the development files of the plasma framework (package name: **libkf5plasma-dev**).