kotelnik/plasma-applet-thermal-monitor #50: Doesn't pick up temperature from a NVMe drive

zell-mbc, 2018-11-09:
"udisks/SM961_NVMe_Samsung".. shows as option but when selected status in the panel is "OFF"
Temperature of my second drive, a regular HDD, shows ok.
OS is Kubuntu 18.10 with all updates installed.

aster94, 2018-11-18:
I was going to open exactly the same issue

@kotelnik could you add something like this?
`sudo nvme smart-log /dev/nvme0 | grep "^temperature"`
@zell-mbc please test this line
or maybe leave people add the possibility to put a command instead or some fixed ones

P.S: great plasmoid!

zell-mbc, 2018-11-18:
Works!
:~$ sudo nvme smart-log /dev/nvme0 | grep "^temperature"
temperature                         : 27 C

aster94, 2018-11-18:
great lets wait for @kotelnik to add this to the plasmoid!

Antoine-Auffret, 2019-06-14:
Here are the full command to get the temperature of an nvme drive in kelvin :
`sudo nvme smart-log /dev/nvme0 --output-format=json | grep -oP '(?<="temperature" : )[^,]*'`
313

The previous command can now be use in the variable UDISKS_TEMPERATURE_CMD_PATTERN to get the temperature of the drive.

How to make this works in the plasmoid ?

zell-mbc, 2019-06-17:
I had a look at the source code.I think the changes need to be made in model-utils.js / function parseUdisksPaths(udisksPaths) but this is well beyond my (limited) programing skills.

Antoine-Auffret, 2019-06-17:
Thanks, I'm not very skilled at JavaScript either but I made some modifications to makes this works for my system. It works **ONLY** if you have one nvme drive "/dev/nvme0" and no other drives.

You need to add to sudoers the nvme command because we need to execute the nvme command with sudo :
`sudo visudo`
Add this at the end of the file :
`your_username ALL=NOPASSWD:/usr/sbin/nvme`

In model-utils.js :

Replacing this line (l.147) :
`var UDISKS_TEMPERATURE_CMD_PATTERN = 'qdbus --system org.freedesktop.UDisks2 {path} org.freedesktop.UDisks2.Drive.Ata.SmartTemperature'`

by these two lines :
`var UDISKS_GREP_TEMPERATURE = '(?<="temperature" : )[^,]*'`
`var UDISKS_TEMPERATURE_CMD_PATTERN = "sudo nvme smart-log /dev/nvme0 --output-format=json | grep -oP '" + UDISKS_GREP_TEMPERATURE + "'"`

Replacing this line (l.157) :
`cmd: UDISKS_TEMPERATURE_CMD_PATTERN.replace('{path}', path),`

by this line :
`cmd: UDISKS_TEMPERATURE_CMD_PATTERN,`

Replacing this line (l.168) :
`return UDISKS_TEMPERATURE_CMD_PATTERN.replace('{path}', UDISKS_PATH_START_WITH + diskLabel)`

by this line :
`return UDISKS_TEMPERATURE_CMD_PATTERN` 

[image]

The main issue is to make the plasmoïd differentiate non nvme drives from nvme drives with two separate commands (one with qdbus and the other with nvme) with the disk label.

We can get the serial number of a drive with this command : `var UDISKS_SERIAL_CMD = 'qdbus --system org.freedesktop.UDisks2 {path} org.freedesktop.UDisks2.Drive.Serial'`

And get the drive path (in this case /dev/nvme0n1) with this command : `lsblk -dpno NAME,TRAN,SERIAL | grep " + UDISKS_SERIAL_CMD + " | grep nvme | awk '{ print $1 }'`

But for the rest, I'm kind of lost for now.