kotelnik/plasma-applet-thermal-monitor #52: QML Type Error

aulospl, 2018-12-14:
The app fails to display temperature automatically, but it works when I hit Reload Temperature Sources.
I'm getting this error from journalctl during startup.
Dec 14 12:12:16 caladan plasmashell[693]: file:///home/aulos/.local/share/plasma/plasmoids/org.kde.thermalMonitor/contents/ui/main.qml:312: TypeError: Type error

System info:
Operating System: Antergos Linux 
KDE Plasma Version: 5.14.4
Qt Version: 5.12.0
KDE Frameworks Version: 5.53.0
Kernel Version: 4.19.8-arch1-1-ARCH
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-7200U CPU @ 2.50GHz
Memory: 7.7 GiB of RAM

roachsinai, 2019-01-19:
Hi @aulospl could you tell me the command you use to get the error message?

cazacugmihai, 2019-09-22:
I have the same error:

```
file:///home/.../.local/share/plasma/plasmoids/org.kde.thermalMonitor/contents/ui/main.qml:312: TypeError: Type error
```

Plasma version: `1.2.8`

Thank you for this awesome widget!

roachsinai, 2019-09-22:
Check PR #54

cazacugmihai, 2019-09-22:
Thank you, @roachsinai!
I confirm that  PR #54 it solves the issue.

roachsinai, 2019-09-22:
Yeah, thanks for feed back.

Focshole, 2019-10-23:
I'm pretty sure i have found the source of the problem. In main.qml:
`property var systemmonitorAvailableSources
    property var systemmonitorSourcesToAdd`
are defined. systemmonitorAvailableSources gets initialized, but systemmonitorAvailableSources doesn't. It then reaches the instruction:
`systemmonitorAvailableSources.push(source)`
 so here comes the TypeError. Simply doing:
`property var systemmonitorSourcesToAdd: []`
fixes the problem. It also adds the missing sources which should load afterwards (no more OFF resources) ;)

roachsinai, 2019-10-23:
@Focshole agree with you, exactly what PR #54 did.