kotelnik/plasma-applet-thermal-monitor #58: I can't choose any lmsensors

EduFrazao, 2019-08-15:
Hi. I'm using this plasmoid, version 1.2.7, but I cant find my CPU Temperature sensor.

I've attached visible options and my sensors command output

[image]

[image]

There is some misconfiguration? How can I do to choose my sensor?

kwhat, 2019-09-22:
This just recently stopped working for me as well.  I have no idea why.

kwhat, 2019-09-28:
Ok, I figured it out for me at least.  It looks like the Gentoo USE flag for lm sensors changed from underscore separated to dash separated.  Ex: `lm_sensors` >> `lm-sensors`

msdobrescu, 2019-12-08:
@kwhat, I have Sabayon, Gentoo based. What is your solution?
I don't see CPU options in the sensors list in this applet, although other tools do.

dwardor, 2019-12-13:
> Ok, I figured it out for me at least. It looks like the Gentoo USE flag for lm sensors changed from underscore separated to dash separated. Ex: `lm_sensors` >> `lm-sensors`

Hey thanks for that ! Solved my similar issue

msdobrescu, 2019-12-14:
What is the solution, then, @kwhat , @dwardor ?

dwardor, 2019-12-19:
Check your use flags (make.conf or package.use) and if you have the old "lm_sensors" change it to the new "lm-sensors".

mleise, 2021-05-12:
The relevant package is `kde-plasma/ksysguard` in case anyone is still wondering in 2021.

blathers123, 2021-06-14:
@mleise Helped me out, thank you!

gchamon, 2021-08-20:
Bug report in the arch linux forums: https://bugs.archlinux.org/task/71237