kotelnik/plasma-applet-thermal-monitor #59: Unusable source selection UI on systems with many sources

phrxmd, 2019-10-30:
On systems with many temperature sources (in my case a Thinkpad X1 with ACPI and `lm-sensors` package), the source selection UI is unusable because the "Group Sources" section of the window grows too much:

[image]

Maybe a list with a scrollbar would be more useful in those cases.

andreacasarin, 2020-04-03:
I'm having the same issue in both X1 Extreme first gen and a Macbook Pro 15. Working on a fix.

andreacasarin, 2020-04-03:
[This](https://github.com/kotelnik/plasma-applet-thermal-monitor/pull/66) works for me, @phrxmd. Could you please let me know if it's working for you aswell?