kotelnik/plasma-applet-thermal-monitor #63: Widget does not start with the system

KOT040188, 2020-01-14:
After starting the system, the widget does not work. I have to turn it on myself, select the data update in the context menu. Previously, he started with the system. Version 1.2.7 KDENeon plasma 5.17.5 https://imgur.com/a/6zftDBf

shmerl, 2020-02-21:
I probably have the same issue. In recent Plasma versions, when you log-in, thermal plasmod shows everything as off. Doing "Reload temperature sources" activates it.

agurenko, 2020-02-25:
You can try to apply this [patch](https://github.com/kotelnik/plasma-applet-thermal-monitor/pull/60/commits/7ecc0888c3575e326c8b75127687ed0a69ff54d7), it was supposed to fix this issue.

shmerl, 2020-02-25:
Thanks, let me try that. How is the project doing? Looks @kotelnik  isn't maintaining it anymore?

agurenko, 2020-02-25:
It hasn't been updated in a while, that's true. I've actually made a fork for myself in pulled changes I need, because patching it manually on a new system got tedious.

penguin86, 2021-11-06:
I can confirm, same issue here. I'll try the one from @agurenko (too bad it isn't included in Debian repositories instead of this one).