kotelnik/plasma-applet-thermal-monitor #64: Project status?

shmerl, 2020-02-24:
There are a bunch of issues and pull requests with fixes that are sitting around for a while. @kotelnik: are you still maintaining the project?

patlefort, 2020-07-25:
Or is there a good alternative? Everything I have tried so far is buggy and crash prone.

alium, 2020-07-25:
I did not find any alternative

use this two patch, with this patchs works wigdet again for me under Qt 5.15:
https://aur.archlinux.org/cgit/aur.git/tree/0001-let-var-systemmonitorAvailableSources-refer-to-an-empty-array.patch?h=plasma5-applets-thermal-monitor-git
https://aur.archlinux.org/cgit/aur.git/tree/0002-Fix-Plasmashell-crash-when-opening-config.patch?h=plasma5-applets-thermal-monitor-git

zhimsel, 2020-07-26:
> Is there any active fork? The author didn't respond in a long time, so he isn't available.

I'd be happy to fork it and maintain it. I don't really know JavaScript that well, but I would be happy to at least review and test PRs.

alium, 2020-07-28:
> > Is there any active fork? The author didn't respond in a long time, so he isn't available.
> 
> I'd be happy to fork it and maintain it. I don't really know JavaScript that well, but I would be happy to at least review and test PRs.

just do it ;-)

agurenko, 2020-07-28:
I've actually [forked it here](https://gitlab.com/agurenko/plasma-applet-thermal-monitor) quite some time ago exactly to merge some patches and not do it manually. It's a gitlab instead of github though but I'm happy to merge and review PRs although my experience is also limited. Also happy to grant permissions for others, so it's not a one man show. Let me know if you're interested

alium, 2020-07-30:
nice @agurenko . i asked for switching to your fork https://bugs.archlinux.org/task/67375

zhimsel, 2020-07-31:
@agurenko I've tested your fork on my system here, and I'm still getting that crash when opening the config window. Using the upstream repo with the patches listed above (see [these two patches](https://aur.archlinux.org/cgit/aur.git/tree/?h=plasma5-applets-thermal-monitor-git&id=ab87c07635be941193df81bc1d404d0ff817236e)), the crash does not occur. I can dig into this more when I have some free time, but otherwise, would you mind investigating and merging that patch in with your fork? I'd be happy to switch the AUR package to use it.

zhimsel, 2020-07-31:
@agurenko Okay, I've managed to merge in those changes to your fork. I'll be creating a PR for you shortly. Once that's merged, I'll switch the AUR package over. I'll also create a BIG BOLD LETTERS issue in this repo to make it easier for people to find your fork when browsing this repo.