kotelnik/plasma-applet-thermal-monitor #67: Applet crashing rear opening Thermal Monitor Settings

hardwareadictos, 2020-05-13:
Rear updating my arch something broke the Widget. Done it on firstly on my laptop and did it two days later on my Desktop (same Arch install
[thermal_monitor_journalctl.txt](https://github.com/kotelnik/plasma-applet-thermal-monitor/files/4621697/thermal_monitor_journalctl.txt)
ation on both) with the same results.

I'm going to upload a log and 

the packets that got updated (because i don't know which one caused the breakage).

[thermal_monitor_journalctl.txt](https://github.com/kotelnik/plasma-applet-thermal-monitor/files/4621702/thermal_monitor_journalctl.txt)

[package_update.txt](https://github.com/kotelnik/plasma-applet-thermal-monitor/files/4621717/package_update.txt)

I already tested other widgets and none of them are having issues.

Let me know if you need more info :)

TimoWilken, 2020-05-14:
I installed `plasma5-applets-thermal-monitor` from the Arch Linux community repository (and nothing else; I had already installed its dependencies). Adding the applet to the plasma bar works, but trying to configure the applet crashes `plasmashell`, even after a reboot.

hardwareadictos, 2020-05-14:
> I installed `plasma5-applets-thermal-monitor` from the Arch Linux community repository (and nothing else; I had already installed its dependencies). Adding the applet to the plasma bar works, but trying to configure the applet crashes `plasmashell`, even after a reboot.

That's my exact issue. I even tried to install it via KDE store and via KDE applet application.

rohitghali, 2020-05-14:
Even I tried installing it from everywhere. Same issue. Best applet. Please fix it.

imatimba, 2020-05-16:
Bug report on KDE's side: https://bugs.kde.org/show_bug.cgi?id=421392

hardwareadictos, 2020-05-16:
> Bug report on KDE's side: https://bugs.kde.org/show_bug.cgi?id=421392

Nice! But I think that's not caused by Nvidia driver update. I suspect more about plasma lts update imo

imatimba, 2020-05-16:
> > Bug report on KDE's side: https://bugs.kde.org/show_bug.cgi?id=421392
> 
> Nice! But I think that's not caused by Nvidia driver update. I suspect more about plasma lts update imo

Yeah, when I saw this bug report just now I did a new comment disregarding the info about de gpu/drivers

Ashark, 2020-05-18:
There is also a [Fan Speed Monitor](https://github.com/dwardor/plasma-applet-fanspeed-monitor) applet, which says in Readme that it is based on this widget, and it also crashes when trying to configure widget.

imatimba, 2020-05-21:
Well, the kde bug report was marked as duplicate of a 4 year old bug report that no one has even look at, it seems.
15 bug reports were marked as duplicates of this old bug report already. Not sure what to think about it. 
It looks like a WONTFIX to me.

hardwareadictos, 2020-05-21:
> Well, the kde bug report was marked as duplicate of a generic 4 year old bug report that no one has even look at, it seems.
> 15 bug reports were marked as duplicates of this old bug report already. Not sure what to think about it.
> It looks like a WONTFIX to me.

Well, that's not a KDE issue, it's an applet issue. We won't get any support outside here...

akien-mga, 2020-05-21:
> Well, that's not a KDE issue, it's an applet issue. We won't get any support outside here...

I beg to differ, if a plasmoid can crash plasmashell, that's a bug in plasmashell.

The issue can likely be worked around in the plasmoid to avoid making plasmashell crash, but the latter should not be crashable by user plugins in the first place.

imatimba, 2020-05-21:
> Well, that's not a KDE issue, it's an applet issue. We won't get any support outside here...

Considering the applet was working fine until now it's possible that the bug is on KDE's side.  
Unless they've changed some internal behavior on purpose that makes code in this applet "the" bug.

imatimba, 2020-05-21:
I was thinking on doing a git bisect, but considering the complexity and size of KDE I wouldn't know where to start.  
The bug could be on plasma-workspace, plasmashell, plasma-framework, etc.

jnines, 2020-05-21:
For the time being I'm just using a command in the Command Output plasmoid to get my CPU/GPU temps.  I don't know if kotelnik is still active or not, or if one of the numerous forks has it fixed.

acutbal, 2020-05-22:
Hi!!
I also have this issue on my computers, every time I try to config the thermal widget crashes plasma.
Regards.

HayriGomer, 2020-05-22:
Same here , latest update broke it , so i think problem is plasma(desktop / framework / QT ... etc) not the applet itself  . Hopefully next update it will be fixed

XinqiBao, 2020-05-23:
same issue here.
And I found another widget `kargos`, can run script and works really well for me.
Thanks @roachsinai shared this [solution](https://github.com/kotelnik/plasma-applet-thermal-monitor/issues/53#issuecomment-457652567), even it was for another issue.

mariocarbone, 2020-05-26:
Is there any way to try to config it via a file?
I removed it and reinstalled apllied the fix and don't work anyway

This is my latest config
[image]

Now i have the shitty icon :(

valdotdev, 2020-05-31:
Same problem on Manjaro 20.0. Crashes when trying to access the settings.

mariocarbone, 2020-05-31:
> Same problem on Manjaro 20.0. Crashes when trying to access the settings.

Do you know how to configure it without menu?

roachsinai, 2020-06-01:
> > Same problem on Manjaro 20.0. Crashes when trying to access the settings.
> 
> Do you know how to configure it without menu?

Should be in `~/.config/plasma-org.kde.plasma.desktop-appletsrc`.

mruiz42, 2020-06-01:
> I beg to differ, if a plasmoid can crash plasmashell, that's a bug in plasmashell.
>
> The issue can likely be worked around in the plasmoid to avoid making plasmashell crash, but the latter should not [...]

Agreed. I 
https://pastebin.com/D0AUaLPT

> I beg to differ, if a plasmoid can crash plasmashell, that's a bug in plasmashell.
>
> The issue can likely be worked around in the plasmoid to avoid making plasmashell crash, but the latter should not [...]

Agreed. Here is the output for mine. It seems to crash right after detecting the disks for me. Not sure what the debug should look like when it opens the config window under normal conditions.
https://pastebin.com/D0AUaLPT

hardwareadictos, 2020-06-05:
Hey! Today i updated my Fedora 32 Workstation and that bug appeared on KDE also.

Beeheerit, 2020-06-09:
I have the same issue here using Arch and Manjaro. It can't be related to any nvidia driver issue, because i don't have any nvidia card nor drivers installed.

BullShark, 2020-06-09:
> I installed `plasma5-applets-thermal-monitor` from the Arch Linux community repository (and nothing else; I had already installed its dependencies). Adding the applet to the plasma bar works, but trying to configure the applet crashes `plasmashell`, even after a reboot.

This issue also affects me in Manjaro. I tried compiling the source from git. I tried installing the AUR package. I had the same problems with both.

bernharl, 2020-06-16:
Same here, on Manjaro. Installed the AUR package that was newly updated. It crashes my desktop when trying to convigure.

jnines, 2020-06-16:
The problem is upstream [https://bugs.kde.org/show_bug.cgi?id=421392](bug report) and who knows if it'll be wontfix.  I've moved on to kargos, requires a little scripting but you do a lot more with it.

moconnor1234, 2020-06-20:
@jnines would you mind sharing the script you used?

Menelkir, 2020-06-20:
In the meantime, here's mine kargos one (I also use [icons-in-terminal](https://github.com/sebastiencs/icons-in-terminal) for the glyphs):

```
cputemp=$(sensors | grep 'Chipset' | awk '{total += $3; count++} END {print total/count}')
gputemp=$(nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader)
sdatemp=$(sudo /usr/bin/hddtemp /dev/sda | cut -c 34-35)
sdbtemp=$(sudo /usr/bin/hddtemp /dev/sdb | cut -c 31-32)

# Status
echo "<font size='2'>  ${cputemp}°&nbsp;  ${gputemp}°&nbsp;¹ ${sdatemp}°&nbsp;² ${sdbtemp}°| font= SF Pro Display:style=Medium size=14"

```

jnines, 2020-06-20:
I use a few different scripts separated out for different things:
```
temp=$(sensors | grep -oP 'Tdie.*?\+\K[0-9.]+') 
echo "${temp%%.*}° | color=red bash='/usr/bin/ksysguard' onclick=bash"
```
for Intel you'd swap Tdie with Package.

```
gpu_temp=$(nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader)
echo "${gpu_temp}° | color=green
```
for Nvidia.  I don't have an AMD GPU but I'm assuming it would show up under lm_sensors.

I also use a few others for things like mic, speakers, battery level for headset.

tkalfaoglu, 2020-07-05:
For my AMD, I used kargos and this command:
sensors|grep edge|cut  -d'+' -f2|cut -d' ' -f1